BleuLlama/TinyBasicPlus #11: ATMega 328p

gregwinn, 2015-05-14:
Hello,
    I have an arduino uno with a ATMega 328p, I am able to load the code to the arduino but when opening the serial window I get the follow:

```
HELLO
TinyBasic Plus v0.13
1039 bytes free.
1024 EEProm bytes total.
OK
>
```

I am trying to run the memory command `MEM` but nothing happens:

```
HELLO
TinyBasic Plus v0.13
1039 bytes free.
1024 EEProm bytes total.
OK
>MEM
```

I try a restart `BYE`:

```
HELLO
TinyBasic Plus v0.13
1039 bytes free.
1024 EEProm bytes total.
OK
>MEMBYE
```

As you see it just adds `BYE` on the end of my last command. Will this just not work on an ATMega 328p or am I doing something wrong?

Thanks for your help!

LjA1210113, 2015-05-14:
SUP! I had this issue too for a while but it's pretty simple, the 328P is waiting for a notification that your done sending a command (ie a line ending) so try changing the drop down menu in the bottom of the serial window that probably says >NO LINE ENDING until it works try >LF AND CF or something first

gregwinn, 2015-05-14:
@LjA1210113 Thank you! That was correct I am using `Both NL & CR` and all works perfect!!

Thanks for the help!

closed 2015-05-14 by gregwinn