BleuLlama/TinyBasicPlus #12: const in order to be put into read-only

gregwinn, 2015-06-09:
I am having an issue with getting this to compile on 1.6.4...

```
variable 'keywords' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

```
I have tried adding 'const' to:
```
static unsigned char keywords[] PROGMEM
```

This does not work
```
const static unsigned char keywords[] PROGMEM
```

Anyone have any idea how to fix this??

bgolab, 2015-06-22:
I used 'const' keyword in front of all variables affected by this issue but new errors appeared. See #13 
-bgolab

bgolab, 2015-06-22:
arduino-1.5.6-r2 - works fine - no compilation errors.
arduino-1.5.7 and onwards is affected. It seems that the issue is related to gcc 4.8.1 introduced in arduino-1.5.7
-bgolab

BleuLlama, 2015-06-22:
ok. i'll check it out

BleuLlama, 2018-06-24:
Fixed with recent releases

closed 2018-06-24 by BleuLlama