BleuLlama/TinyBasicPlus #13: arduino 1.6.5 - variable 'xxx' must be const in order to be put into read-only section

bgolab, 2015-06-22:
When using Arduino: 1.6.5 (Windows 7), Board: "Arduino Uno" (gcc 4.8.1 was introduced in arduino 1.5.7BETA):
TonyBasicPlus.ino:278:33: error: variable 'keywords' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
TonyBasicPlus.ino:370:33: error: variable 'func_tab' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
TonyBasicPlus.ino:385:31: error: variable 'to_tab' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
TonyBasicPlus.ino:390:33: error: variable 'step_tab' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
TonyBasicPlus.ino:395:34: error: variable 'relop_tab' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
TonyBasicPlus.ino:415:36: error: variable 'highlow_tab' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
variable 'keywords' must be const in order to be put into read-only section by means of '__attribute__((progmem))'

After using 'const'keyword the following errors appear:
TinyBasicPlus-arduino1.6.5.ino:776:23: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:476:13: error:   initializing argument 1 of 'void scantable(unsigned char*)' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino: In function 'short int expression()':
TinyBasicPlus-arduino1.6.5.ino:905:22: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:476:13: error:   initializing argument 1 of 'void scantable(unsigned char*)' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino: In function 'void loop()':
TinyBasicPlus-arduino1.6.5.ino:1144:21: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:476:13: error:   initializing argument 1 of 'void scantable(unsigned char*)' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:1382:21: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:476:13: error:   initializing argument 1 of 'void scantable(unsigned char*)' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:1390:23: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:476:13: error:   initializing argument 1 of 'void scantable(unsigned char*)' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:1677:26: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]
TinyBasicPlus-arduino1.6.5.ino:476:13: error:   initializing argument 1 of 'void scantable(unsigned char*)' [-fpermissive]
invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]

bgolab, 2015-06-22:
arduino-1.5.6-r2 - works fine - no compilation errors.
arduino-1.5.7 and onwards is affected. It seems that the issue is related to gcc 4.8.1 introduced in arduino-1.5.7

bgolab, 2015-06-22:
Anyway, it's duplicate of #12

bgolab, 2015-06-22:
https://code.google.com/p/glcd-arduino/issues/detail?id=57
suggests to use 'const' keyword but as I described earlier some new errors appear. So it's a partial solution for our particular case.
-bgolab

BleuLlama, 2015-06-22:
understood.  I've taken care of this for other projects... I'll dive in and
figure it out.

BleuLlama, 2018-06-24:
This has been fixed with recent versions of TBP.

closed 2018-06-24 by BleuLlama