BleuLlama/TinyBasicPlus #17: FOR loop with top limit of 32767

gabrias, 2016-05-09:
I don't know if that is a error inherent to basic language in general, but when i use the value 32767 as the top limit for the FOR loop, the loop will run forever since the variable will go from 32767 to -32768 and keep running.

drawkula, 2018-04-25:
This is not a typical feature of all BASICs... ;-)

The addition in https://github.com/BleuLlama/TinyBasicPlus/blob/083bba9237770f05b3971f2ff5b1a2a51dcd5530/Arduino/TinyBasicPlus.ino#L1501 overflows into the sign bit.

For the same reason...
```
10 FOR I=-32760 TO -32768 STEP -1
20 PRINT I
30 NEXT I
```
...runs into positive numbers instead of terminating at `-32768`.

BleuLlama, 2018-06-24:
Closed,  TinyBasic functions as designed.  If this is an issue, open a new ticket to cover overflows or whatever.  This is pretty standard for integer basics.

closed 2018-06-24 by BleuLlama