BleuLlama/TinyBasicPlus #19: TVout libraries causing problems

kipperklank, 2016-07-17:
when I include tye TVout library, not even using it just including, i get the error: 'unsigned char* sp' redirected as different kind of symbol

even if I get this fixed, how,  would I output the console to TV?

kipperklank, 2016-07-17:
man this libray has been dormit for a loooong time, I doubt I will get a responce

BleuLlama, 2016-07-17:
Hi. I haven't used the library for a long time now.  There are much better
ways to accomplish this same task with newer versions of the OS, since at
least iOS 5.x.

Build the example code to see how to use it.  It will create a window into
which you can add things like you do with the main window.

-s

BleuLlama, 2016-07-17:
OH. wait. sorry I got confused about which library this email was
pertaining to.  let me get back to you.

kipperklank, 2016-07-17:
lol and wow that was fast

BleuLlama, 2016-07-17:
Okay. The version of the TV Out library that you may be using might be
colliding with one of the definitions in this code.  Be aware that without
TV Out, this code will use most of your AVR's ram... you'll have just a few
hundred bytes for your program space.  The TV out library needs at least a
K to be useful.

I would not recommend using it, and I believe I pursued it at one point,
but it was experimenting with it and found it to not be a good way to go.

One thing you could do is have a second arduino (or propeller, or whatever)
as a dedicated video display unit, then use serial or I2C to connect the
two together.  Grant Searle has a project here:
http://searle.hostei.com/grant/MonitorKeyboard/index.html  which might be a
nice interface to this.

Someone has also hacked the serial interface chip, the AT32u4 (iirc) on the
newer Arduino Unos to be a video display driver.  here's a link to that
project, which happens to be an Apple II emulator here:
http://dpeckett.com/turning-the-arduino-uno-into-an-apple  To be fair
though, this project is pretty hardcore. ;)

I hope that this helps you.  I just know that trying to add TV Out library
to the Tiny Basic code base could be difficult to infeasible, sadly.

Cheers, and let me know how it goes!

-s

kipperklank, 2016-07-17:
yes im using  a mega2560 for this and i think it has enough recorces sence i have a 2560 mega and an uno that might be a good idea to use two
and also as i try to use the SAVE command the terminal says unimplemented, i thought i could use sd csrds for file in/output?

kipperklank, 2016-07-17:
Im thinking about using the Wire libraries? how would i do that? I want to "dump" the serial terminal of the Mega 2560 running basic to the Uno running the tvout library.

BleuLlama, 2016-07-18:
Wire is essentially I2C, so you'd hook up the SCL and SDA pins of the mega
(I2C Bus) with ground i think, to the same lines on the Uno, which I
believe are A4 and A5.

Here's an example of someone doing it:
http://www.instructables.com/id/I2C-between-Arduinos/

Then have the Mega be the Master, and the Uno be the slave, i would think,
with the TV Out library on the Uno.

I've never used this myself, although there are some projects where I need
to learn I2C comms between arduinos.

The Wire library itself was pretty straightforward to get going.  I've used
it to talk with a clock chip, and that was super easy.

The great thing about it is that it's low overhead since it's basically an
interface to existing hardware.  So it won't use many resources on either
'duino.

-s

kipperklank, 2016-07-18:
I did it and it works, funny thing is that I dont even have to use I2C libraries or edit code. if I put the serial monitor example in the TV out and change the baud rate to match tye BASIC board (mega) and connect the TX and RX respectively  it outputs the serial from the BASIC board to the TVout board (uno) and it displays the serial. I don't have to even edit ether of the sketches wow

BleuLlama, 2016-07-18:
Awesome! :D

For my little basic machine I ended up getting a 1.8" color LCD module from
Adafruit and did something similar.  I used that as the display device
(9600 baud in, text display out) and a commodore keyboard interfaced to
another arduino which spit out 9600 baud serial as the input device.  the
two together acted like a serial terminal.

-s

kipperklank, 2016-07-18:
i cant implement the ps2keyboard libraries beacuse the ide that i use to compile the basic one
the old ide doesnt support renaming libraries
ex : PS2Keyboard  keyboard;
and if i rename keyboard to PS2Keyboard
i get the error: PS2Keyboard does not have a name type and i dont knowbtye old way to do this nor what its called so googleing it is chalenging

kipperklank, 2016-07-18:
typo : "...dont know the old way to..."

BleuLlama, 2018-06-24:
Closing this as the issue has been resovled. 👍

closed 2018-06-24 by BleuLlama