BleuLlama/TinyBasicPlus #20: Ps2 keyboard libraries?

kipperklank, 2016-07-18:
how would i use a ps2 keyboard for this?
Is is already implemented and I don't know it?

BleuLlama, 2016-07-18:
It is not already implemented in the current codebase... as far as I can
remember...

You need to put the PS2 keyboard on a serial UART of the micro.  for
example:
http://playground.arduino.cc/Main/PS2Keyboard

I'd get that working on its own, then integrate it into the main codebase.

-s

kipperklank, 2016-07-18:
neat and thanks if you do

BleuLlama, 2016-07-18:
Oh. i meant that as a recommendation. sorry that came out weird.

I would recommend that if you go that route, that you experiment with that
and get it working on its own, then integrate it into the main application
from there.

that's what i meant to say. ;)

-s

kipperklank, 2016-07-18:
ok ill e let you know if it works

BleuLlama, 2018-06-24:
Any update on this, @kipperklank?