BleuLlama/TinyBasicPlus #25: Would adding 23LC1024 increase the SRAM, 39 bytes free

spairo1741, 2017-06-21:
After compilling TinyBasicPlus  i received a message
" HELLO
TinyBasic Plus v0.15
39 bytes free.
1024 EEProm bytes total.
OK
>mem
39 bytes free.
1024 EEProm bytes total.
877 EEProm bytes available.
>"
Adding 23LC1024 increase the SRAM ?

BleuLlama, 2017-06-21:
Sure.  You'll have to rewrite the memory accesses to work through it
though.

If you need more space, turn off some of the options (SD interface, beeper
etc)

vitasam, 2018-01-30:
Hi,
I just recently found this project. I have (slow progressing) project - an [Arduino-Nano shield with 8x5 keyboard, 16x2 LCD and 23LC1024 SRAM](https://github.com/vitasam/Basic-Nano).

I hope that it will be possible to fit TinyBasic and LCD driver in to atmega368p. And it will be great to modify memory access to support 128Kb SRAM chip.

vitasam, 2018-02-16:
Found this one: 
https://github.com/rhelmus/virtmem

BleuLlama, 2018-06-24:
As mentioned in another ticket, the overhead with using a virtual memory library is great. You'd lose the 39 bytes you have free before gaining access to the new RAM! :(

Also, the code itself is kinda messy with regard to expanding it such that it uses external RAM for the program and command line.

There has been discussion of adding SPI or I2C calls to the language to accomodate SPI ram for general storage.

BleuLlama, 2018-06-24:
Closing this ticket in favor of tickets #41  and #42 which cover adding the functionality mentioned here.

closed 2018-06-24 by BleuLlama