BleuLlama/TinyBasicPlus #26: virtmem library support

leonardoelias300, 2017-07-30:
It would be great if you use the virtmem library from github.com/rhelmus/virtmem so you can increase ram memory using SD card.

BleuLlama, 2017-07-30:
That is really super nifty! I hadn't heard of that one before but I've had ideas on similar lines myself. I'll put this on my list of things to check out and see what can be done. 

I only really have an SD interface to work with right now, but I'll check out the other memory options there!

Thanks for this suggestion!!!
S

Sent from your iPhone.

BleuLlama, 2017-08-04:
Working on it...

BleuLlama, 2018-01-20:
I've looked through it and there appears to be a lot of overhead for using that library.  TBP needs all the resources it can get.  I can make a very stripped down version of SD virtual memory support -- i experimented with this quite a bit -- but it would require restructuring the TBP code a lot, and I sadly just don't have time to do this.  (as witnessed by this issue as being from July.)

It's a really great idea and may happen, but not super soon.

vitasam, 2018-02-18:
It will be really great to add also SPI SRAM support. This is my experiment with mentioned above virtmem project:
```
  char testbuf[] = "This is a test string buffer\nThis is a test string buffer\nThis is a test string buffer\nThis is a test string buffer\nThis is a test string buffer\nThis is a test string buffer\nThis is a test string buffer";
    
    // Allocate a char buffer in virtual memory and store the address to a virtual pointer
    VPtr<char, SPIRAMVAlloc> str = valloc.alloc<char>(128000);    
    VPtr<char, SPIRAMVAlloc> ptr = str;

    memset(str, 'A', 65000);
    Serial.print("Read from str[0]: "); Serial.println(str[0]);
    char a= *(ptr + 10);
    Serial.print("char a= *(ptr + 10): "); Serial.println(a);

    *(ptr + 100000) = 'H';
    char h = *(ptr + 100000);
    Serial.print("char h = *(ptr + 100000): "); Serial.println(h);

    str[60000] = 'B';
    Serial.print("Read from str[60000]: "); Serial.println(str[60000]);

    str[126000] = 'C';
    Serial.print("Read from str[126000]: "); Serial.println(str[126000]);

    memcpy(&str[127200], testbuf, sizeof(testbuf));
    memset(testbuf, 0, sizeof(testbuf));
    memcpy(testbuf, &str[127200], sizeof(testbuf));
    Serial.print("Read from str[127200]: "); Serial.println(testbuf);

    valloc.free(str); // Free the virtual memory
```

Result:
**Read from str[0]: A
char a= *(ptr + 10): A
char h = *(ptr + 100000): H
Read from str[60000]: B
Read from str[126000]: C
Read from str[127200]: This is a test string buffer
This is a test string buffer
This is a test string buffer
This is a test string buffer
This is a test string buffer
This is a test string buffer
This is a test string buffer**

BleuLlama, 2018-02-19:
Yes. I've experimented with these libraries and they work quite well.  My problem is that the way that Tiny Basic deals with memory is a bit messy with relation top re-using the input buffer for storage and such.  I experimented a bit with it but wasn't really getting anywhere at the time.

vitasam, 2018-02-19:
Hi,
Ok, I got it.

What do you think about other way of using SPI SRAM - adding a new keyword (function) for reading and writing to SRAM? Could be useful for some usecases when one will read e.g. ADC value and want to store it somewhere for post-processing

BleuLlama, 2018-06-24:
Sounds reasonable to me.  I do not have such a device to test with. General SPI or I2C commands might be nice to have as well, as a foundation to things like this.  Then we can read from real sensors from our BASIC programs as well.

I'll gladly implement it if i have the devices to test with.  (Lend me a shield with the device, etc).

BleuLlama, 2018-06-24:
Closing this in favor of #41 and #42 which will cover this functionality

closed 2018-06-24 by BleuLlama