BleuLlama/TinyBasicPlus #30: The input command has no documentation and I can not get it to work

pcmr106, 2017-11-30:
I have been working on this for about an hour now, I would like to know how to use the input command

BleuLlama, 2017-11-30:
As mentioned in the docs the INPUT command is currently nonfunctional 

Sent from your iPhone.

BleuLlama, 2018-06-24:
Hi. Not sure if you're still interested in this, but INPUT is implemented now. (it might have been in the past as well.)  But as you mention; no documentation.

I have documented it on the main page.  In short:  10 INPUT A    - will take input from the user, and store it in variable A.  It will accept numbers/expressions.  So you can use integers "43", "88" or variable names "a"   etc But not textual strings.

closed 2018-06-24 by BleuLlama

pcmr106, 2018-07-12:
Thanks, I'll have to try it out now