BleuLlama/TinyBasicPlus #33: Added functionality to TinyBasic

Volhout, 2018-02-16:
dear BeluLama,

I added (over the last years) functionality to your tinybasic.
Maybe you want to look at it and add this to your build (I am not planning my own repository).
Please find the code at: https://drive.google.com/open?id=1SSq9poOlyginnRclIGtSJ7WTLhNCP80p

Added:
- servo support on UNO pin 9 and 10 (used PWMServo lib because it is not very memory hungry).
  "SERVO pin, value" , and "NOSERVO pin"
- keyword abbreviation. Makes code less readable, but more compact.
  i.e. "DWRITE" can be abbreviated to "DW. And "PRINT" to "P."

Regards,

Volhout (Harm)
deleeuw52@gmail.com

Volhout, 2018-06-04:
dear BlueLlama,

I have updated your tinybasic again, and added some commands for robotics.
[https://github.com/Volhout/TinyBasicPlus](url)

Regards,

Volhout

BleuLlama, 2018-06-24:
Hello. I finally have gotten to this ticket. Sorry about the delay. 

I really like your implementation of the SERVO command/code/  I'll integrate it soon.  

One thing; I'd appreciate it if you change your version numbering somehow.  Your project is based on 0.13 i believe?  Please name it accordingly... 0.13.volhout.1 or something.  please try to keep the mainline version numbers as the primary x.xx  versions, to prevent confusion in the future.  (ie, if someone has problems and sees yours as the "newest"... all of a sudden, you've gotta start supporting the whole thing.  :o )

Volhout, 2018-06-24:
Dear Scott,

In the mean time I have added more functionality to TinyBasicPLus.
I have also fixed a bug in in the servo command.

My version is now at 0.17t (t for test, but it is confirmed to fix the bug in re-assigning timers)
There is also an explanation of the commandset in TinyBasic (word document), maybe nice to copy that.

Please search Github for TinyBasicPlus and Volhout (my name on Github).
I would love to see (most of the) the changes incorporated in your version since it is quite popular. But got no response until now, and decided to post it.
But I must say that I did not have the opportunity to test my version on anything except Arduino (UNO mostly).

I taylored this version for a guy called Richard, who is using it in Arduino MIni's for an educational project for poor children, that they can build a simple SUMO bot, sponsored from the community. He has help from local people to turn ultra cheap chineese servo's and stuff into kits for these kids.

My changes all centre around adding functionality that does not consume awful lots of RAM since on the UNO it is scarce.
I added:
HELP, VARS, MIN, MAX, MILS, appart from the SERVO and NOSERVO that you have seen.
And I fixed a few bugs.

The MIN and MAX are not implemented as  functions, but could be changed to functions if that is more in style of the language.
I looked at the way you implemented the INPUT command.
Reason was  to make the BASIC text as brief as possible.

I also re-arranged the order in which commands are in the tables, to make sure that frequently used commands can be abbreviated more.

Please look at my changes, and see if you patch some of these in your code.

I may have violated a few rules in C and C++, please look and correct, since I have not a lot of experience in C. I have difficulty using pointers especially, so some of my code could be improved dramatically, I am sure.

I have also minimized the buffer size in the hardware serial buffers, in the CPP file, since the different versions of Arduino compiler use different names for that buffer size (I use ancient 1.03 on linux, but the new 2018 release on Windows). I think you should not implement that change for the masses, since it only gives 64 more bytes RAM, but could prove a problem with each Arduino toolchain update.
The 2018 Arduino version has a far better compiler (code size is about 10k) as the old (code size is 14k), but both use about equal RAM.....

If, in the future you would spend time on TinyBasicPLus, the things I see most rewarding would be
- store line numbers as INT, not in ASCII, that saves RAM, and should be easy to implement. Every \n is followed by an INT.
- add a second number to LIST (i.e. LIST 10,30 to list lines 10,20,30).
- see if you can cut the amount of RAM used by the interpreter. I have no idea where it all goes, but more than 1k is consumed.

It has been an exiting effort, rewarding, and I learned a lot.
Thank you for this chance....

Regards,

Volhout / Harm

guruthree, 2018-06-25:
> - store line numbers as INT, not in ASCII, that saves RAM, and should be easy to implement. Every \n is followed by an INT.

Do you mean line numbers at the starts of lines or line numbers used as part of GOTO GOSUB etc? Having just spent some time playing with how the code is stored in memory, line numbers at the start of lines are already stored as short ints.

(From [line 1114](https://github.com/BleuLlama/TinyBasicPlus/blob/d319264b6c50eb8ba36e7365f57e7945ee558805/TinyBasicPlus/TinyBasicPlus.ino#L1114) where the current input is pared, `linenum = testnum();` reads in the line number but also moves the txtpos pointer forwards, to where characters start on the current line. Subsequently, *txtpos is moved back by three to overwrite the ASCII number with the short int number and the number of characters of text that make up the line.)