BleuLlama/TinyBasicPlus #35: [FYI] TinyBasic Plus v0.14 on ESP8266

drawkula, 2018-04-24:
Nothing dramatic, just a "for your information" issue:

TBP-0.14 compiles without changes in Arduino for ESP8266:
```
TinyBasic Plus v0.14
2801 bytes free.
OK
>_
```
After change of `#define RAMEND 4096-1` to `#define RAMEND 32768-1`:
```
TinyBasic Plus v0.14
31473 bytes free.
OK
>_
```
A more dynamic way to set `RAMEND` probably is known somewhere... ;-)

It coughs on "INPUT" and some other stuff (I haven't logged them) with a reboot but simple stuff like...
```
>10 print "moo"
>run
moo
```
...works.

Maybe some ESP8266 fans will have fun looking closer at this?
On the other hand, there is a full blown ESP-BASIC out there.

Well... it's just an info...

BleuLlama, 2018-04-24:
Seems about right. ;)

The free ram space was always a bit of a hack in TBP.  You can't consume *ALL* available RAM for your use as the system gets very unstable.  You have to leave at least 125ish bytes free reported by the compiler otherwise print statements, function calls, etc will just hang or reboot the AVR without notice.

The free space defines in the code were tested guesses -- a balance of program space vrs real "free" memory without the micro crashing.

I've not done any work with the ESP chips... although I do believe that TBP was included with the ESP32...

BleuLlama, 2018-06-23:
Adding in #ifdef ESP8266 to cover this.  with 1.15

Sidenote: the reason that probbaly breaks is that RAMEND is not "the complete size of memory" but rather "the amount of free heap space we most likely have while this is running".  So it really should be 16k - (estimation of space for libraries used and stack).  If you set it for 16k, it WILL crash. :D 

Will set it at 8192 for now, but without an ESP8266 platform to directly test on, this is a wild guess.

BleuLlama, 2018-06-23:
Oh my bad on that. it IS the memory space, but for 8266 the computations are likely set differently. so yeah, still setting it at 8k.

closed 2018-06-23 by BleuLlama