BleuLlama/TinyBasicPlus #36: [FAQ?] Please explain the license changes in TBPs family tree

drawkula, 2018-04-25:
TBP is [indirectly](http://hamsterworks.co.nz/mediawiki/index.php/Arduino_Basic) based on [[Gordon Brandly]'s 68000 Tiny Basic](https://web.archive.org/web/20170306113457/http://members.shaw.ca:80/gbrandly/68ktinyb.html) with this copyright:
```
******************************************************************
*                                                                *
*               Tiny BASIC for the Motorola MC68000              *
*                                                                *
* Derived from Palo Alto Tiny BASIC as published in the May 1976 *
* issue of Dr. Dobb's Journal.  Adapted to the 68000 by:         *
*       Gordon Brandly                                           *
*       12147 - 51 Street                                        *
*       Edmonton AB  T5W 3G8                                     *
*       Canada                                                   *
*       (updated mailing address for 1996)                       *
*                                                                *
* This version is for MEX68KECB Educational Computer Board I/O.  *
*                                                                *
******************************************************************
*    Copyright (C) 1984 by Gordon Brandly. This program may be   *
*    freely distributed for personal use only. All commercial    *
*                      rights are reserved.                      *
******************************************************************
```
Is it really safe to use the MIT license for derived ports?

(This really is a question only, not an accusation!)

BleuLlama, 2018-04-25:
The code that I started with had no license attached to it.  By default, my code in github defaults to MIT license.  If you'd like, please feel free to diff my code with what was provided by Michael Field (which ostensibly still carries the above referenced copyright (not a license)) and apply the MIT license to my additions.  There's not really a 1:1 mapping of this 1976/1984 copyright notice-to-modern-license equivalent.  One of the Creative Commons licenses could come close...

BleuLlama, 2018-06-23:
Your "laws i don't know" comment was quite snarky, but I get your point. To counter your snark: I don't have much time to work on projects these days, and I've decided today that I should work on figuring out this license thing that no one else will really care about, notice, or follow.... rather than working on implementing features in this project that people have requested.  ;)

I felt that MIT license was reasonable in satisfying the intent of the original and Mike's adaption of the project.

Walking forward from the original... (Just to document it all here; I know you already know all of this, @drawkula )

The "40 years old journal publication" Dobbs version's license is not an issue.  Its license is essentially "personal use only".  However, Gordon Brandly's 68000 version was a reimplementation of the syntax, so the Dobbs license is not applicable. The 68k license however might be.

Fields' effort was taking the 68000 asm, and reimplementing it in C.  It is a "derivative work", surely, but does it mean the original license applies?  Probably.  His was also provided without license, freely on the internet, where it remains, without license of any kind.  From my conversations with him, I believe his intent was to provide this freely, for people to work on, putting it firmly in public domin/open source land. However commercial distribution of it is unclear.  I've contacted him to get his opinion.

Now that all said, it already has been included with commercial products, namely the ESP32 which included a version of TinyBasicPlus on the rom it shipped with.  Neither myself nor Mike were contacted about this to my knowledge.

twoporylyj, 2021-09-26:
But is it a derivative work at all? Studying assembly code, determining how it works, and reimplementing the same in C from scratch. That's how Compaq reimplemented IBM BIOS from scratch, and it was not considered a derivative work.