BleuLlama/TinyBasicPlus #53: is there a Way to connect an 1602/2004 screen to an Arduino running Tinybasic and showing everything on it?

ChaosLeo07, 2020-12-28:
Hello! 
I just found my first Arduino Uno and found out that i have put TinyBasic on it. And i like it!!
I just went through the code and think that the stuff gets written to the Serial monitor with serial read/write. So, now to the question. Is it possible to display the stuff from the Serial Monitor to an LCD? i think i should be able to use lcd.write instead of Serial.read/write. Will that be possible?
Best Regards
ChaosLeo07

BleuLlama, 2020-12-28:
That should work yes.  Just be aware that the lcd library will likely use memory in the arduino, which is already quite tight. 

Sent from your iPhone.

ChaosLeo07, 2020-12-28:
> That should work yes.  Just be aware that the lcd library will likely use memory in the arduino, which is already quite tight. 
> 
> Sent from your iPhone.
> 
> > 
> > 
> > Hello!
> > I just found my first Arduino Uno and found out that i have put TinyBasic on it. And i like it!!
> > I just went through the code and think that the stuff gets written to the Serial monitor with serial read/write. So, now to the question. Is it possible to display the stuff from the Serial Monitor to an LCD? i think i should be able to use lcd.write instead of Serial.read/write. Will that be possible?
> > Best Regards
> > ChaosLeo07

so if i would take an arduino with more ram, i will get better results?