BleuLlama/TinyBasicPlus #54: GOSUB on ESP8266 crashes

ivangalysh, 2021-02-09:
I am trying to get TinyBASIC plus working the on the ESP8266 to be used by some kids for a robotics camp. Everything seems to work except for the gosub command. The processor crashes every time it is used.

10 print "test"
20 gosub 100
30 goto 10
100 print "gosub"
110 return

This is my test program. Has anyone come across this issue?
On another note, I did go through the source code and modify the section for using SD cards to use SPIFFS. 

Ivan

BleuLlama, 2021-02-09:
Can you try the original on your ESP8266, without SPIFFS or even SD support?
Do you get any warnings or errors when you build it in either configuration?

(On some platforms, you can enable SD support, but it uses a ton of RAM,
resulting in there not being enough space for the program, or stack, or it
just corrupts itself when it runs...)

ivangalysh, 2021-02-09:
I downloaded the source again and compiled. There are no warnings or errors. GOSUB still crashes.  FOR also crashes.

Here is the dump from the ESP8266:
Exception (9):
epc1=0x40201d51 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffefe8b depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 0190
3fffff70:  3ffee34c 3ffee34e 3ffee308 40201d1e  
3fffff80:  3fffdad0 3ffefdfe 3ffefeb8 40201640  
3fffff90:  3fffdad0 00000000 3ffefeb8 3ffeff24  
3fffffa0:  3fffdad0 00000000 3ffefee4 40202a8c  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100c25  
<<<stack<<<

What I did on the version I modified, I created a 512 unsigned long array for a stack and a different stack pointer just for GOSUB and FOR. The FOR command started working but GOSUB still isn't. I suspect the issue with with word alignment.
I suspect word alignment since the program array is byte array and the stack may not be 32-bit aligned when a GOSUB is called.

ivan

ivangalysh, 2021-02-09:
The exception I get is for unaligned write or read. I suspect that is due to the stack operation of trying to move 32 bit pointers into an 8 bit array and not being 32-bit aligned. I'm going to modify the code to copy the pointer values one byte at a time. I suspect that will work.

Ivan

ivangalysh, 2021-02-10:
The exception is a memory alignment error. The problem is the stack where you save the structures for GOSUB and FOR and retrieve for RETURN and NEXT. I modified the code to save the pointers one byte at a time instead of writing the 32-bit address value. It now works. I also enabled the SD card and changed it to SPIFFS and that works too. I now have what I need for a robot kit for kids.

If you want to see what I did, I can send you my version. I just need to go back and add comments.
Thanks
Ivan

closed 2021-02-10 by ivangalysh

BleuLlama, 2021-02-10:
I'd love to see the changes for that fix, and for your SPIFFS additions too.

ivangalysh, 2021-02-12:
Here is my modified version of the code so it works properly on the ESP8266. I replaced the SD card with LittleFS which replaces the SPIFFS library. I also added commands to delete files and rename files. 

Thanks for the software, I plan to use this for educational outreach efforts on a robot kit I designed to be used by grades 5-6.

[TinyBasicPlus_esp.ino.zip](https://github.com/BleuLlama/TinyBasicPlus/files/5972550/TinyBasicPlus_esp.ino.zip)